IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #22: Arduino - Ejemplo Control Monitor Serie - Grupos: Movilidad y Control

Martii3141, 2025-04-11:
Copiar y pegar el siguiente codigo en TinkerCad y ejecutarlo.
Obetivo de la Guia: Analizar las nuevas funciones del monitor serie que son presentadas (read, parseInt, available, etc). Observar funciones creadas por el usuario y su invocacion.
Actividad: Hacer un seguimiento de la variable actividadActual y poder saber en todo momento cual es su valor.

`
//Guia Monitor Serie Arduino

//Declaracion de vbles globales		Las Variables Globales son variables que se declaran fuera del loop, del setup y de cualquier otra funcion creada por el usuario. Dichas variables se les conoce como globales, ya que se pueden invocar en cualquiera de las otras funciones
String nombre = "";			//Los tipos de datos String son una concatenacion de caracteres.
int actividadActual;
int edad;
int respuesta;
String despedida = "chau";
bool tutorial = true;

void setup(){
	Serial.begin(9600);		//Esta linea sirve para inicializar el monitor serie. Sin esta linea, no podremos usar tal herramienta. El argumento de 9600, nos indica la tasa de transferencia de datos. No es el unico valor que puede tener de transferencia de datos, pero por ahora solo usaremos este.
	Serial.println("******************************************************");	//Las funcion print y println del monitor Serial, nos sirven para imprimir texto o valores de vbles. La unica diferencia entre print y println es que println agrega un renglon al final de todo el texto impreso en dicha linea.
	Serial.println("* Bienvenidxs a la guia del Monitor Serie de Arduino *");
	Serial.println("****************************************************** \n");	//Otra forma de indicar que queremos un salto de renglon es poniendo \n en donde deseamos que se produzca dicho salto.
	Serial.println("Para continuar, envia como te llamas desde el monitor serie.");
	while(Serial.available() == 0){			//La funcion available verifica cuantos caracteres hay disponibles para ser leidos por la maquina. Es decir que en un momento inicial no hay caracteres para ser leidos (Serial.available = 0), una vez que enviamos un mensaje Serial.available() deja de ser 0, ya que tiene caracteres por leer aun. 
		
	}
	nombre = Serial.readString();		//La funcion read lee los caracteres disponibles que hay en el monitor serie, y en este caso, los guarda en la variable nombre. Al leer los caracteres que estaban disponibles, Serial.available(), volvera a ser 0, ya que los datos que estaban disponibles ya fueron leidos con la funcion de esta linea.
	Serial.print("\n Hola ");
	Serial.print(nombre);
	Serial.println(".");
}

void loop(){
	mostrarConsigna();
}

void mostrarConsigna(){
	Serial.print("---- Actividad Nro ");
	Serial.print(actividadActual + 1);
	Serial.println(" ----");
	switch(actividadActual){			//La estructura switch case, nos sirve para asignar bloques de codigo segun valores especificos de una sola variable. Es decir, le damos una variable, la cual tomara distintos valores, y para cada valor, ejecutara una secuencia de lineas de codigo especificadas. La sintaxis es la presentada en el ejemplo. Para mas informacion, ingresar a la pagina oficial de arduino.
		case 0:					//Si la variable vale 0, entonces se ejecuta todo lo de abajo, hasta break;
		  Serial.println("Ingresa tu edad: ");
		  procesamientoEntrada();
		  break;
		case 1:					//Si la variable vale 1, entonces se ejecuta todo lo de abajo, hasta break;
		  Serial.println("¿Tu edad es un numero par o impar? \t \t 1 - Impar \t 2 - Par");
		  Serial.println("Ingresar el numero de la respuesta que corresponda (1 o 2)");
		  procesamientoEntrada();
		case 2:					//Si la variable vale 2, entonces se ejecuta todo lo de abajo hasta break;
		  Serial.println("Finaliza esta guia escribiendo y enviando 'chau' al monitor serie.");
		  procesamientoEntrada();
		  break;
	}
return;
}

void procesamientoEntrada(){
	while(!tutorial){
		
	}
	while(Serial.available() == 0){

	}
	switch (actividadActual){
		case 0:
		  edad = Serial.parseInt();			//La funcion parseInt sirve para leer un dato entero ingresado por el monitor serie. A diferencia de read, este nos lee el dato como lo ingresamos y no en el sistema ASCII.
		  Serial.print("Tu edad es: ");
		  Serial.println(edad);
		  if(edad >= 18){
			  Serial.println("Sos mayor de edad.");
		  }
		  else{
			  Serial.println("Sos menor de edad.");
		  }
		  actividadActual++;		//Esta linea le suma 1 a la variable actividadActual. Es una forma compacta de escribir actividadActual = actividadActual + 1. Se puede hacer lo mismo con una variable para restarle 1, intercambiando el ++ por --.
		  mostrarConsigna();
		  break;
		case 1:
		  respuesta = Serial.parseInt();
		  if(respuesta == 2 && edad % 2 == 0){
		  	Serial.println("Tienes razon. Tu edad es numero par.");
		  }
		  else if(respuesta == 1 && edad % 2 == 0){
			Serial.println("Respuesta incorrecta. Tu edad si es un numero par.");
		  }
		  else if(respuesta == 1 && edad % 2 != 0){
		  	Serial.println("Respuesta correcta. Tu edad es un numero impar.");
		  }
		  else if(respuesta == 1 && edad % 2 == 0){
			Serial.println("Tienes razon. Tu edad es un numero impar.");
		  }
		  actividadActual++;
		  mostrarConsigna();
		  break;
		case 2:
		  if(Serial.readString() == despedida){
		  	Serial.println("El tutorial ha finalizado.");
			tutorial = !tutorial;
		  }
	}
return;
}

`

sebalander, 2025-05-03:
Marcar si ya lo completaron:

- [ ] @ASH101-CODE Compeltado
- [ ] @Leandro284-AR Completado
- [ ] @srmartbr17 Completado

sebalander, 2025-05-07:
ya todos lo completaron asi que cierro el issue

closed 2025-05-07 by sebalander