IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #25: logica para saber el angulo de la pelota con respecto al eje de la camara

rafaelmoya2010, 2025-04-15:
[image]

rafaelmoya2010, 2025-04-15:
h azul es distinta de h negra

sebalander, 2025-04-16:
esta bien, les queda como tarea (usen este mismo issue) para construir la logica, tipo pseudocodigo, o por lo menos escrita en palabras en castellano, de lo que tiene que hacer el algoritmo que calcula. recuerden:

- Inputs: las coordenadas x,y de la pelota en pixeles, son coordenadas en la imagen
- Output: cuantos radianes debe girar el robot para que la pelota quede centrada en la imagen
- Parametros: d? k? esto definanlo uds

sebalander, 2025-05-03:
hola gente. @3liasCo    @rafaelmoya2010     @roma1keta . No estoy seguro pero me parece que quiza esto ya quedo resuelto, no? o sea, recuerdo que (corrijanme si me equivoco) ya estan sabiendo como hacer la cuenta y ya estan trabajando para tenerla andando en la OpenMV, no? si es asi, pueden cerrar este issue tranquilamente. saludos

3liasCo, 2025-05-03:
Ya lo estamos usando, así que creo que ya no sirve tener la lógica bien escrita

closed 2025-05-03 by 3liasCo