IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #26: Python Basico e Introduccion a Objetos

Martii3141, 2025-04-16:
## **Google Colab de Python basico**

En resumen este documento de Google les muestra como es la sintaxis basica de Python con las diferentes funciones como el _if... else_, _while_, etc. 

https://colab.research.google.com/github/carlosalvarezh/Pensamiento_Algoritmico/blob/main/Modulo04_EstructurasControlFlujo.ipynb#scrollTo=7DZQloSLuTYb

## **Google Colab de Programacion Orientada a Objetos (POO) en Ingles**

Este documento es del mismo formato que el anterior, solo que este tutorial está en ingles. Si pueden leerlo, entenderlo y resolverlo genial. Sino esta el siguiente link que es documentacion en español.

https://colab.research.google.com/github/QuantEcon/lecture-python-programming.notebooks/blob/master/python_oop.ipynb

## **Tutorial POO en español**

Este tutorial es de la pagina ellibrodepython.com que esta buena por si quieren chequear mas cosas de python y seguir aprendiendo por su cuenta. En esta ocasion la vamos a usar para aprender lo basico de Programacion Orientada a Objetos.

https://ellibrodepython.com/metodos-estaticos-clase-python

## ✅ Seguimiento de ejercicios

| Ejercicio                      | @3liasCo | @rafaelmoya2010  | @roma1keta |
|-------------------------------|----------|----------|----------|
| Python Basico   | <ul><li>[ ] </li></ul>    | <ul><li>[ ] </li></ul>    | <ul><li>[ ] </li></ul>      |
| Python POO   |<ul><li>[ ] </li></ul>     |  <ul><li>[ ] </li></ul>      |<ul><li>[ ] </li></ul>      |

sebalander, 2025-04-23:
# ✅ Seguimiento de ejercicios @3liasCo
 - [ ] Python Basico
 - [ ] Python POO

sebalander, 2025-04-23:
# ✅ Seguimiento de ejercicios @rafaelmoya2010
 - [ ] Python Basico
 - [ ] Python POO

sebalander, 2025-04-23:
# ✅ Seguimiento de ejercicios @roma1keta
 - [ ] Python Basico
 - [ ] Python POO

closed 2025-05-06 by roma1keta