IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #27: Tarea para la Casa: Corregir Codigo Ejemplo Monitor Serie

Martii3141, 2025-04-16:
Como se habran dado cuenta, el codigo que les pasé, al finalizar el tutorial, les pide nuevamente que envien 'chau' por el monitor serie y luego de eso no hace mas nada el codigo.
La actividad que tienen es corregir dicho error; cuando envien 'chau', lo ultimo que tiene que mostrar el monitor serie es "El tutorial ha finalizado."

## **Recomendaciones**
- Traten de primero entender porque ocurre lo del codigo original. Para esto tienen que poder entender la gran mayoria del codigo que esta escrito.
- Consulten las partes del codigo que no entiendan, que les generen duda de porque se hizo de tal forma, etc.
- Modifiquen el codigo o hagan otros codigos de prueba para tratar de entender mejor las funciones que se estan usando.

## 
**Las dudas, como asi tambien el codigo corregido mandenlo por los comentarios de este issue.**
##
| Ejercicio                      | @Leandro284-AR | @srmartbr17 | @ASH101-CODE  |
|------------------------------|----------|----------|----------|
| Codigo Corregido       |  <ul><li>[ ] </li></ul>    |   <ul><li>[ ] </li></ul>      |   <ul><li>[ ] </li></ul>      |

sebalander, 2025-04-23:
✅ Seguimiento de ejercicios @Leandro284-AR
- [ ] Codigo Corregido

sebalander, 2025-04-23:
✅ Seguimiento de ejercicios @srmartbr17
- [ ] Codigo Corregido

sebalander, 2025-04-23:
✅ Seguimiento de ejercicios @ASH101-CODE
- [ ] Codigo Corregido

Leandro284-AR, 2025-04-27:
// Declaración de variables globales
String nombre = "";
int actividadActual;
int edad;
int respuesta;
String despedida = "chau";
bool tutorial = true;

void setup() {
  Serial.begin(9600);
  Serial.println("******************************************************");
  Serial.println("* Bienvenidxs a la guia del Monitor Serie de Arduino *");
  Serial.println("****************************************************** \n");
  Serial.println("Para continuar, envia como te llamas desde el monitor serie.");

  while (Serial.available() == 0) {
    // Esperar nombre
  }

  nombre = Serial.readString();
  Serial.print("\n Hola ");
  Serial.print(nombre);
  Serial.println(".");
}

void loop() {
  if (tutorial) {
    mostrarConsigna();
  } else {
    // Detiene el loop completamente para que no haga nada más
    while (true);
  }
}

void mostrarConsigna() {
  Serial.print("---- Actividad Nro ");
  Serial.print(actividadActual + 1);
  Serial.println(" ----");

  switch (actividadActual) {
    case 0:
      Serial.println("Ingresa tu edad: ");
      procesamientoEntrada();
      break;
    case 1:
      Serial.println("¿Tu edad es un numero par o impar? \t \t 1 - Impar \t 2 - Par");
      Serial.println("Ingresar el numero de la respuesta que corresponda (1 o 2)");
      procesamientoEntrada();
      break;
    case 2:
      Serial.println("Finaliza esta guia escribiendo y enviando 'chau' al monitor serie.");
      procesamientoEntrada();
      break;
  }
}

void procesamientoEntrada() {
  while (!tutorial) {
    // No hacemos nada si el tutorial ya terminó
  }

  while (Serial.available() == 0) {
    // Esperar entrada
  }

  switch (actividadActual) {
    case 0:
      edad = Serial.parseInt();
      Serial.print("Tu edad es: ");
      Serial.println(edad);
      if (edad >= 18) {
        Serial.println("Sos mayor de edad.");
      } else {
        Serial.println("Sos menor de edad.");
      }
      actividadActual++;
      break;

    case 1:
      respuesta = Serial.parseInt();
      if (respuesta == 2 && edad % 2 == 0) {
        Serial.println("Tienes razón. Tu edad es número par.");
      } else if (respuesta == 1 && edad % 2 == 0) {
        Serial.println("Respuesta incorrecta. Tu edad sí es un número par.");
      } else if (respuesta == 1 && edad % 2 != 0) {
        Serial.println("Respuesta correcta. Tu edad es un número impar.");
      } else if (respuesta == 2 && edad % 2 != 0) {
        Serial.println("Respuesta incorrecta. Tu edad no es un número par.");
      }
      actividadActual++;
      break;

    case 2:
      String entrada = Serial.readString();
      entrada.trim();  // Quita espacios y saltos de línea
      if (entrada == despedida) {
        Serial.println("El tutorial ha finalizado.");
        tutorial = false;
      } else {
        Serial.println("No entendi eso. Por favor, escribe 'chau' para salir.");
      }
      break;
  }
}

closed 2025-05-22 by Leandro284-AR