IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #36: Cortar el panel lego y definir material es para adherir a Zircon

sebalander, 2025-05-01:
Al final de la clase del martes falto que:

# definan donde cortar el panel amarillo

**Objetivo: indicar a @sebalander como cortar el panel amarillo de Legos. Las indicaciones deben ser un diagrama (dibujo, imagen, etc) bien claro indicando donde cortar y ademas pasar en texto aparte las medidas de donde se deberia cortar. por favor si tienen dudas o no saben como hacer: escriban, pregunten, etc**. Contesten en este issue un comentario con ese contenido.

Seguramente necesiten buscar informacion en internet sobre el panel. les paso una ayuda por si les sirve:  "yellow lego Technic, Panel Plate 11 x 19" 

[image]

tambien supongo que necsitaran informacion sobre la base movil: "Robomov Zircon"

# Definir materiales para adherir/fijar el panel amarillo a la base Zircon

Objetivo: proponer una lista de materiales que servirian para fijar el panel amarillo de lego a la base de zircon. Comenten este issue liestando qué materiales necesitarian.

sebalander, 2025-05-03:
ok, gracias. el lunes lo busco.

no se olviden de los materiales para adherir

lauri-an, 2025-05-06:
Un material para adherir al zircon podría ser poxipol (?
Para cortar el lego dejamos una marca al costado

sebalander, 2025-05-06:
ok. conseguiremos poxipol y algo para cortar el lego.

lauri-an, 2025-05-06:
El panel lego ya esta cortado y ya tenemos el poxipol

closed 2025-05-06 by lauri-an