IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #37: Pregunta del grupo OpenMV al grupo de Movilidad

sebalander, 2025-05-01:
En clase me plantearon en el grupo OpenMV que necesitaban consultar algo al grupo de Movilidad. Dejo este issue en blanco para que se estriban entre ustedes y hagan la consulta. Usen este Issue como un chat/hilo de intercambio pero que a la vez sera el registro de como se reolvio el asunto.

grupo **OpenMv**:
 - @3liasCo 
 - @rafaelmoya2010 
 - @roma1keta 

grupo **Movilidad**:
 - @Vale15-tb 
 - @mijaelb78 
 - @vickyT049

3liasCo, 2025-05-01:
Hola! Queria saber cómo tenía que ser el ángulo que necesitan para que se maneje el robot. 
El cuadro azul de la imagen es lo que ve la cámara, nosotros estamos haciendo que calculé en radianes el ángulo que tiene la pelota respecto al eje de la camara, si el ángulo esta a la izquierda del eje va a ser positivo, si está a la derecha del eje va a ser negativo. 
[image]

vickyT049, 2025-05-06:
0 equivale a sentido horario y 1 equivale a sentido antihorario

vickyT049, 2025-05-06:
para poder movernos en grados nosotros usamos regla de tres simple ,osea tomamos un valor como referencia y de ahí cualquier ángulos es multiplicado por el tiempo y dividido por el grado de referencia que elegimos ej:

 #include <Arduino.h>
#include <zirconLib.h>
float angulo;

float girar (float angulo) {
  
  
  float tiempo = angulo * 0.075 / 90;
  
  motor1(100, 1);
  motor2(100, 1);
  motor3(100, 1);
  delay(tiempo);
  motor1(0, 1);
  motor2(0,1);
  motor3(0,1);
  delay(2000);
  return tiempo;
}
void setup(void)
{
  Serial.begin(9600);
  InitializeZircon();

}

void loop(void)
{   
 angulo=60;   
 Serial.print("el angulo que gira es: " );
 Serial.println(girar(angulo));
}

closed 2025-05-26 by 3liasCo