IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #39: Tutorial Diseño 3D

Martii3141, 2025-05-02:
# 🧊 Primeros Pasos en Diseño 3D con Tinkercad

Vamos a empezar a explorar el mundo del diseño 3D usando Tinkercad.

Para dar nuestros primeros pasos, mira este video introductorio. Te enseñara lo basico para empezar a moverte en la plataforma:

https://www.youtube.com/watch?v=csvBRWfgf1I&ab_channel=ElRinc%C3%B3ndePrimaria

---

# 🛠️ Actividad: Diseñando un Disco con Agujeros

Ahora que viste el video y exploraste un poco Tinkercad, tu primera mision de diseño es sencilla:

1.  **Crea un disco:** Usa las formas basicas para hacer un cilindro plano que parezca un disco. Que no mida mas de 1 cm de alto.
2.  **Haz dos agujeros:** Usa las formas "huecas" para hacer dos agujeros circulares pequeños en alguna parte de tu disco.
3.  **Agrupa todo:** Asegurate de agrupar el disco y los agujeros para que se conviertan en una sola pieza.

¡Animate a probar las herramientas basicas que viste en el video para lograrlo!

---
- [ ] @mijaelb78 
- [ ] @rafaelmoya2010
- [ ] @vickyT049

sebalander, 2025-05-07:
@mijaelb78 cargue el model stl "Smashing Jofo.stl" en el software para procesarlo antes de mandar a la impresora 3D y me lo toma como que tiene 2cm de diamtro el disco:

[image]

asi que antes de imprimir lo voy a escalear para que tenga 10cm de diametro.

mijaelb78, 2025-05-07:
Hola profe yo lo hice pero no me deja haberlo al que descargue aller,ya lo voy a hacer en mi notebook y se lo mando

closed 2025-05-30 by sebalander