IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #41: Especificación de la Función "MovimientoLineal"

sebalander, 2025-05-07:
## Objetivo

Implementar una función en Arduino para controlar el movimiento lineal del robot en una de las tres direcciones definidas por los ejes de sus tres motores.

## Nombre de la Función

Copiar y pegar esta plantilla de coigo y luego completar el contenido de la funcion.

```cpp
/**
 * @brief Controla el movimiento lineal del robot en la dirección de uno de sus tres ejes.
 * 
 * @param eje Número del eje (1, 2 o 3) que define la dirección de movimiento.
 *        - 1: Movimiento en la dirección del primer motor.
 *        - 2: Movimiento en la dirección del segundo motor.
 *        - 3: Movimiento en la dirección del tercer motor.
 * 
 * @param distancia Distancia a recorrer en centímetros. Puede ser positiva ("hacia donde apunta" la rueda del eje seleccionado) o negativa (dirección opuesta).
 * 
 * @note Factor de conversión entre distancia y tiempo de giro ... ESPECIFICAR ACA XXX
 */
void MovimientoLineal(int eje, float distancia){
  // aca deben completar con su codigo
  // activar motores, delays, etc
}
```

## Comportamiento Esperado

1. Según el valor del parámetro "eje", la función debe definir qué motores activar y en qué sentido.
2. La dirección de giro de cada motor debe depender del signo de "distancia":

   * Distancia positiva: Movimiento en dirección del eje.
   * Distancia negativa: Movimiento en dirección opuesta.
3. Los motores deben activarse durante un tiempo proporcional al valor absoluto de "distancia". Ustedes deben definir el factor de conversión entre distancia (cm) y tiempo (ms) experimentalmente y explicitarlo en su código.

## Recomendaciones

* Utilicen un sistema de funciones auxiliares si lo consideran necesario (por ejemplo, para activar motores con sentido y velocidad específicos).
* Definan el factor de conversión como una constante al inicio del script, de modo que sea fácil de ajustar.
* Comenten adecuadamente el código para explicar su lógica.

sebalander, 2025-05-28:
veo el commit https://github.com/IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025/commit/d346c5cf93bff2cf32a2fe2d26a9df823abd8848 que ya parece que esta mas cerca de terminarse la tarea, pero falta.

sebalander, 2025-06-04:
[Draft] 
@vickyT049 estamos mirando el codigo que subio @Vale15-tb  recien en el commit https://github.com/IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025/commit/852f60ffdfc7c725dab4cf56046461df535bd3fe. veo por ejemplo esto:

```c
float girar (float tiempo) {

tiempo = angulo * 0.075 / 90;
motor1(100, 1);
motor2(100, 1);
motor3(100, 1);
delay(girar);
motor1(0, 1);
motor2(0, 1);
motor3(0, 1);
delay(2000);

return tiempo;
}
```

trato de entender que quisieron  hacer. les hago algunas preguntas, como para que me expliquen y entender un poco su idea. quiza asi entienda como ayudarlas mejor

1. el input de la funcion `girar` es un tiempo:
```c
float girar (float tiempo) {
```

Martii3141, 2025-06-12:
@vickyT049 la ultima clase estuvimos charlando sobre como hacer las funciones correctamente con sus repectivos argumentos y demas. Pudiste avanzar algo? Seguis teniendo dudas de como hacer algo con ese codigo? Por favor subilo al codigo asi vemos que hiciste, tambien si tenes alguna duda sobre como seguir con el codigo preguntanos asi te ayudamos.

sebalander, 2025-06-25:
@vickyT049 si entendi bien, todavia tenes para trabajar la funcion de giro. trata de avanzar en tu casa asi traes algo ya para testear el martes. si necesitas ayuda avisa.