IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #45: Mas diseño 3D

Martii3141, 2025-05-08:
# Diseño 3D: Prototipos Iniciales en Tinkercad

Actividades de diseño en Tinkercad para empezar a crear prototipos para nuestro robot.
Cada uno hagan una actividad. Yo les asigne de una manera. Si quieren cambiarse la actividad entre ustedes no hay problema. La idea es que las tres actividades queden hechas, no importa por quien.
---

**Actividad 1: Estructura de 3 Pisos**

Diseñar en Tinkercad una estructura basica que permita ensamblar tres niveles o pisos para el robot. Piensen en como unirian las bases de cada piso.

- [ ] @vickyT049  

---

**Actividad 2: Pieza Compatible con Lego**

Diseñar en Tinkercad una pieza que se pueda ensamblar con bloques de Lego. Puede ser una base con orificios o una pieza con espigas que coincidan aproximadamente con las dimensiones de los Legos para poder atornillarla o pegarla a otras partes.

- [ ] @rafaelmoya2010 

---

**Actividad 3: Soporte de Camara con Inclinacion**

Diseñar en Tinkercad un soporte para la camara que permita cambiar su angulo de inclinacion. Piensa en un mecanismo sencillo que permita ajustar la posicion de la camara hacia arriba o hacia abajo.

- [ ] @mijaelb78 

---

sebalander, 2025-05-30:
primera idea, construir en tinkercad un primer prototipo donde el arco  en realidad no sea curvo, sino que sean mas bien columnas rectas, que arriba se unan con una viga horizontal, que seria la manija.

vickyT049, 2025-06-03:
https://www.tinkercad.com/things/aD3iHP7unZN-estructura-para-los-pisos/edit?returnTo=https%3A%2F%2Fwww.tinkercad.com%2Fdashboard
primer diseño de las estructuras

Martii3141, 2025-06-04:
@vickyT049  estamos tratando abrir el link con seba y no nos esta dejando, fijate si nos lo podrias pasar por whatsapp al link porfis asi mandamos a imprimir el stl

[image]
ese error nos tira cuando abrimos el link

sebalander, 2025-06-04:
@vickyT049 como tarea para el martes que viene seria que avances con mejoras en el diseño 3D. si tenes ideas compartilas por aca y las discutimos. si tenes algun STL donde plasmaste una idea, subilo asi lo puedo abrir yo en 3D y verlo.

Martii3141, 2025-06-12:
@vickyT049 como va el diseño de la estructura? hubo alguna novedad con eso esta semana? Si tenes algun archivo para mandar, pasalo porfis asi lo hacemos imprimir y lo testeamos lo antes posible. Si todavia no hay archivo, me gustaria ver una fotito por lo menos de como va la cosa jaja, me da curiosidad como va quedando esa idea que estaba genial cuando la vi

sebalander, 2025-06-23:
@vickyT049 ... pasanos lo que sea que tengas. dejanos ver tu diseño, asi te podemos ayudar. Lo mejor seria si podes subir el STL a la carpeta de diseño 3D. 
el link de tinkercad que pasaste antes lo intente abrir pero me sigue dando error.

sebalander, 2025-06-23:
@mijaelb78 pudiste hacer algo del diseño de soporte de camara que me habias comentado?

mijaelb78, 2025-06-23:
Hola profe si logré hacer una pared más alta para la cámara con orificios para atornillarla y la bace es de 15 cm y lo de la pared sería de 3 cm de altura

sebalander, 2025-06-23:
buenisimo! aca si queres ir investigando las dimensiones de esos agujeros, hay que buscar en la pagina del fabricante de la OpenMV. ahi tienen que tener diagramas detallados de las dimensiones. 

fijate aca: https://openmv.io/products/openmv-cam-h7-plus

[image]