IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #58: control modo arquero

ASH101-CODE, 2025-05-13:
[image]

sebalander, 2025-05-16:
@ASH101-CODE avisame como venis, si necesitas algo. 

minimamente proba de meter el codigo que tenes en tinckercad y empeza a probarlo. ponele prints para que imprima informacion de lo que esta calculando en el serial monitor asi podes ir entendiendo lo que hace.

ASH101-CODE, 2025-05-20:
// Constantes
const float ANGULO_REFERENCIA = 10.0;      // grados
const float DISTANCIA_REFERENCIA = 50.0;   // cm

// Función que calcula el desplazamiento lateral
float calcularDesplazamiento(float anguloGrados, float distanciaCm) {
  // Convertir ángulo a radianes para usar tan()
  float anguloRad = radians(anguloGrados);

  // Calcular desplazamiento lateral (positivo = derecha, negativo = izquierda)
  float desplazamiento = distanciaCm * tan(anguloRad);

  return desplazamiento; // en centímetros
}

void setup() {
  Serial.begin(9600);
  Serial.println("Modo arquero iniciado.");
}

void loop() {
  // Simulamos inputs de ejemplo (podés reemplazar esto por valores reales)
  float anguloInput = 10.0;      // en grados
  float distanciaInput = 50.0;   // en cm

  // Calcular desplazamiento
  float desplazamientoLateral = calcularDesplazamiento(anguloInput, distanciaInput);

  // Mostrar resultado
  Serial.print("Ángulo: ");
  Serial.print(anguloInput);
  Serial.print("°, Distancia: ");
  Serial.print(distanciaInput);
  Serial.print(" cm → Desplazamiento lateral: ");
  Serial.print(desplazamientoLateral);
  Serial.println(" cm");

  delay(1000); // Esperar 1 segundo antes de repetir
}