IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #59: para de investigación para el profe

vickyT049, 2025-05-13:
@sebalander buscar como conectar la teensy al Arduino para poder programar

sebalander, 2025-05-16:
@vickyT049  te comento lo que estoy probando en mi compu portatil (creo que es igual que la tuya asi que deberiamos poder lo mismo)

# instalar la IDE de arduino
nada que decir, es facil

# Seguir el instructivo dettensyduino para instalar en la IDE de Arduino la libreria teensyduino

https://www.pjrc.com/teensy/teensyduino.html

# identificar que Tennsy tenemos, fijate en la propia placa:

[image]aca dice que es teensy 4.1

[image]este detalle del modelo del chip a veces suele ser util para identificar con mucha precision que teensy tenemos

alternativamente hay otras formas de identificar, hay tutoriales que te hacen contar la cantidad de pines, por ejemplo.

la porpia Ide de Arduino ayuda a veces, por ejemplo aca la detecta y la lista al final del desplegable:

[image]

# seleccion de la Teensy 4.1

[image]

# probar algun ejemplo

la IDE trae ejemplos precargados

[image]

sebalander, 2025-05-22:
resuelto @vickyT049 ? si ya esta, cerra el issue