IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #60: Transformacion en perspectiva

sebalander, 2025-05-16:
Para estudiar teoria sobre transformacion en perspectiva. la idea es tomar coordenadas pixeles (2D) y calcular con eso coordenadas en cm en el espacio fisica (el plano 2D)

Ver este ejemplo y explicacion:
https://www.youtube.com/watch?v=y1EgAzQLB_o

Esta son las dos funciones que se van a utilizar mas abajo:
- https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga4abc2ece9fab9398f2e560d53c8c9780
- https://docs.opencv.org/3.4/d2/de8/group__core__array.html#gad327659ac03e5fd6894b90025e6900a7

### **Actividad Práctica: Transformación de Perspectiva (Google Colab)**

#### **Materiales:**

* Hoja de papel (A4 o cualquier tamaño conocido).
* Cámara (puede ser la de un celular).
* Cuenta de Google y acceso a Google Colab.

---

#### **1. Preparación del Entorno:**

* Coloca la hoja de papel sobre una superficie plana (mesa o piso).
* Dibuja una cruz en algún lugar dentro de la hoja.

#### **2. Captura de la Imagen:**

* Toma una fotografía de la hoja desde un ángulo inclinado, simulando la vista que tendría la cámara de tu robot (mirando hacia abajo y en ángulo).
* Asegúrate de que en la imagen se vean claramente las cuatro esquinas de la hoja y la cruz dibujada.
* Sube la imagen a Google Colab.

---

#### **3. Configuración de Google Colab:**

* Abre un nuevo notebook en Google Colab.
* Instala y carga las librerías necesarias:

```python
# Instalar OpenCV (si es necesario)
!pip install opencv-python-headless -q

import cv2
import numpy as np
import matplotlib.pyplot as plt
```

---

#### **4. Cargar y Mostrar la Imagen:**

```python
from google.colab import files
from PIL import Image

# Subir la imagen
uploaded = files.upload()
image_path = list(uploaded.keys())[0]

# Cargar la imagen
imagen = cv2.imread(image_path)
imagen_rgb = cv2.cvtColor(imagen, cv2.COLOR_BGR2RGB)

# Mostrar la imagen
plt.imshow(imagen_rgb)
plt.title("Imagen Original")
plt.show()
```

---

#### **5. Identificación de Coordenadas en la Imagen:**

* Identifica a ojo las coordenadas de las cuatro esquinas de la hoja en la imagen (en píxeles).
* Anota estos cuatro pares de coordenadas (u, v).
* Identifica también las coordenadas de la cruz dibujada en la imagen (u, v).

```python
# Coordenadas en la imagen (u, v) - ajustar estos valores a tu imagen
pts_img = np.float32([
    [150, 200],   # Esquina 1
    [500, 180],   # Esquina 2
    [480, 600],   # Esquina 3
    [130, 620]    # Esquina 4
])

# Coordenadas de la cruz en la imagen (u, v)
pts_cruz = np.float32([[320, 400]]).reshape(-1, 1, 2)
```

---

#### **6. Definición de Coordenadas Físicas:**

* Define las coordenadas físicas reales de las esquinas de la hoja.
* Para una hoja A4 (21 cm x 29.7 cm), puedes usar:

```python
# Coordenadas físicas (x, y) en cm
pts_fisicas = np.float32([
    [0, 0],      # Esquina 1
    [21, 0],     # Esquina 2
    [21, 29.7],  # Esquina 3
    [0, 29.7]    # Esquina 4
])
```
cuidado! las esquinas tienen que estar referidas  a las esquinas que se anotaron en pixeles, en el mismo orden se den listar las coordenadas de las esquinas.
---

#### **7. Cálculo de la Matriz de Homografía:**

```python
# Calcular la matriz de homografía
H, _ = cv2.findHomography(pts_img, pts_fisicas)
```

---

#### **8. Transformación de Coordenadas (Cruz):**

```python
# Aplicar la transformación a la cruz
pts_cruz_fisicas = cv2.perspectiveTransform(pts_cruz, H)

print("Coordenadas físicas de la cruz:", pts_cruz_fisicas[0][0])
```

---

#### **9. Visualización de las Coordenadas Físicas:**

```python
# Visualizar las coordenadas físicas de las esquinas y de la cruz
plt.figure()
plt.scatter(*zip(*pts_fisicas), color='blue', label='Esquinas (cm)')
plt.scatter(pts_cruz_fisicas[0][0][0], pts_cruz_fisicas[0][0][1], color='red', label='Cruz (cm)')
plt.title("Coordenadas Físicas Calculadas")
plt.xlabel("X (cm)")
plt.ylabel("Y (cm)")
plt.grid(True, linestyle='--', alpha=0.6)
plt.legend()
plt.show()
```

---

#### **Resultado Esperado:**

* Verás una gráfica mostrando las coordenadas físicas de las esquinas de la hoja (en azul) y la cruz (en rojo).
* El cálculo de las coordenadas de la cruz se habrá realizado automáticamente usando la homografía.

---

¿Te interesa que simplifique la identificación de coordenadas (u,v) pidiéndole al usuario hacer clic directamente en la imagen para capturarlas automáticamente?

3liasCo, 2025-05-17:
https://youtu.be/ITrexC2MA0c?si=b_uKcWzlCcVmAvrd

sebalander, 2025-05-22:
@3liasCo hacer este issue es tu tarea. anda siguiendolo paso por paso. no es facil, no va a salir de una todo junto, pero anda probando y avisame como venis. ok? asi te doy una mano

3liasCo, 2025-05-28:
b'\xff\xd8\xff\xe0\x00\x10JFIF\x00\x01\x01\x01\x00\x00\x00\x00\x00\x00\xff\xfe\x00\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xdb\x00C\x00\x10\x0b\x0c\x0e\x0c\n\x10\x0e\r\x0e\x12\x11\x10\x13\x18(\x1a\x18\x16\x16\x181#%\x1d(:3=<9387@H\\N@DWE78PmQW_bghg>Mqypdx\\egc\xff\xdb\x00C\x01\x11\x12\x12\x18\x15\x18/\x1a\x1a/cB8Bcccccccccccccccccccccccccccccccccccccccccccccccccc\xff\xc0\x00\x11\x08\x00\xf0\x01@\x03\x01!\x00\x02\x11\x01\x03\x11\x01\xff\xc4\x00\x1f\x00\x00\x01\x05\x01\x01\x01\x01\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x01\x02\x03\x04\x05\x06\x07\x08\t\n\x0b\xff\xc4\x00\xb5\x10\x00\x02\x01\x03\x03\x02\x04\x03\x05\x05\x04\x04\x00\x00\x01}\x01\x02\x03\x00\x04\x11\x05\x12!1A\x06\x13Qa\x07"q\x142\x81\x91\xa1\x08#B\xb1\xc1\x15R\xd1\xf0$3br\x82\t\n\x16\x17\x18\x19\x1a%&\'()*456789:CDEFGHIJSTUVWXYZcdefghijstuvwxyz\x83\x84\x85\x86\x87\x88\x89\x8a\x92\x93\x94\x95\x96\x97\x98\x99\x9a\xa2\xa3\xa4\xa5\xa6\xa7\xa8\xa9\xaa\xb2\xb3\xb4\xb5\xb6\xb7\xb8\xb9\xba\xc2\xc3\xc4\xc5\xc6\xc7\xc8\xc9\xca\xd2\xd3\xd4\xd5\xd6\xd7\xd8\xd9\xda\xe1\xe2\xe3\xe4\xe5\xe6\xe7\xe8\xe9\xea\xf1\xf2\xf3\xf4\xf5\xf6\xf7\xf8\xf9\xfa\xff\xc4\x00\x1f\x01\x00\x03\x01\x01\x01\x01\x01\x01\x01\x01\x01\x00\x00\x00\x00\x00\x00\x01\x02\x03\x04\x05\x06\x07\x08\t\n\x0b\xff\xc4\x00\xb5\x11\x00\x02\x01\x02\x04\x04\x03\x04\x07\x05\x04\x04\x00\x01\x02w\x00\x01\x02\x03\x11\x04\x05!1\x06\x12AQ\x07aq\x13"2\x81\x08\x14B\x91\xa1\xb1\xc1\t#3R\xf0\x15br\xd1\n\x16$4\xe1%\xf1\x17\x18\x19\x1a&\'()*56789:CDEFGHIJSTUVWXYZcdefghijstuvwxyz\x82\x83\x84\x85\x86\x87\x88\x89\x8a\x92\x93\x94\x95\x96\x97\x98\x99\x9a\xa2\xa3\xa4\xa5\xa6\xa7\xa8\xa9\xaa\xb2\xb3\xb4\xb5\xb6\xb7\xb8\xb9\xba\xc2\xc3\xc4\xc5\xc6\xc7\xc8\xc9\xca\xd2\xd3\xd4\xd5\xd6\xd7\xd8\xd9\xda\xe2\xe3\xe4\xe5\xe6\xe7\xe8\xe9\xea\xf2\xf3\xf4\xf5\xf6\xf7\xf8\xf9\xfa\xff\xda\x00\x0c\x03\x01\x00\x02\x11\x03\x11\x00?\x00fisH\xc2\xc2\xe6\x97u\x03\xb0\xa0\xd2\xe6\x80\x174\xa0\xd01sK\xba\x80\x00is@\xc5\xcd&h\x10\xb9\xa4&\x90\xc6\x93M&\x80\x1aM&h\x01\xacj64\x01\x0c\x86\xa0cHd-0\x07\x93PM/\x1c\x1a\xceZ\x9a\xc0\xa8\xeeMR\x94\xfc\xd4\xe2:\x84tV\x86!E\x00!\xa4\xa0aE0\x14t\xa0R\x01\x0fZ3L\x0e\xe74\xb9\xa0\xcd\x0bK\x9a\x06(\xa5\xa4\x01K\x9a\x00PisE\xc6(<\xd2\xe6\x80\x174P\x01Hh\x01\r0\x9a@4\x9ai<\xd0\x021\xa8\xda\x81\x90\xc8j\xbb\x9a\x00\xcd\xb9r\x1a\xa0\xdeMC4\x88\x13\xc5U\x97\xefQ\x1d\xc72:+C \xa2\x80\x03IL`)qH\x03\xb5%0\x03I@\xce\xea\x81H\xc8ZZ\x06-- \x16\x8ac\x14R\xd2\x01GZZ\x00QE0\nJ@!\xa6\x9a\x00i\xa6\x1a\x00i\xa6\x1aC!\x92\xabI\xde\x81\x99\xb7_x\xd4\x15,\xb8\x8azUY>\xf5\x11\t\x8c\xa2\xb43\n(\x004\x94\xc6\x02\x9c\x0e\r \x15\xf1\x8c\x8a\x8e\x84\x08(\xa63\xba\xef@\xe9H\xc8ZZ\x06-\x14\x0cZZB\x16\x96\x98\xc5\x14\xa2\x90\x0bK@\tA\xa0\x06\x9akR\x01\x86\x9ah\x01\xa6\x98h\x19\x0c\x95ZC\xc1\xa0\x0c\xcb\x93\x96\xa8*Y\xa4E\xedU\xa4\xfb\xd4Ds\x19Ehd\x14\xb4\x00\x86\x92\x81\x86qK@04\xda`\x82\x8a\x06wTR2B\xd2\xd01h\xa0b\xd1\x9a@(4\xea\x00)h\x01E-\x00-%\x004\xd3\x1b\xad&\x03I\xa6\x9a\x00cS\t\xa0d2\x9a\xab\'CI\x8d\x19\xb7\x1fz\xa1\xa4\xcb\x88v\xaa\xf2}\xea"\x13\x1bIZ\x19\x85\x14\x00\x1aJ\x06\x14\xaa}h\x00"\x92\x80BQLgu\x9eh\xfaR2\x01K@\xc34P\x02\xd2\xd2\x18\xb4\xea\x00ZQ@\x85\xa5\x14\x80ZCL\x06\x9e\x94\xc3Hc\x1a\x98M\x004\xd3\x18\xd02\x19*\xac\x9d)\r\x19\xd7\x1fz\xa1\xa9e\xc4^\xd5Z@wS\x88Lm\x15\xa1\x98Q@\x08i(\x18R\xe6\x80\x17\xb54\xd0\x81\tK\x8e)\x8c\xee(\xa5c0\xa5\xcd\x00\x19\xa5\xcd\x03\nQ@\x85\x06\x9d\x9a@(\xa7\n\x06(\xa5\xa4 \xa44\x00\xd2j6<\xd0\x03\t\xa6\x93@\xc6\x9aa\xa0d2t\xaa\xd2\x1e)1\xa3:\x7f\xbdQT\xb2\xe2\x07\xa5"H\x80\xe1\xc58\x84\xc8f\xdb\xbf\xe4\xe9L\xe2\xac\xccCE\x00!\xa4\xa61qI@\x0fNE4\x8ey\xa0B\x803\xd6\x9e\x9f)\xecE\x00vF\x8a\t\x0c\xd1\xed@\xc5\xa2\x80\xea-\x14\x80p\xa7\n\x04(\xa7\x03H\x05\xcd.h\x00\xcd4\x9a\x00c\x1a\x8c\x9a\x064\x9ai\xa0\x044\xc64\x86C\'J\xac\xfd\r\x0c\x11\x9f0\xcbqH\x90;\xf4\x15,\xd1\x16\xa3\xb1$|\xd4\x92i\x8a\xdd\r8\x8aN\xe4\r\xa5\xb7f\xa8\xdbM\x94t\xe6\xac\x826\xb1\x98\x7f\rFm\xa5\x1f\xc2h\x00\x1098 \xd4\xcbg\x91\xcd\x005\xec\xd8}\xda\xae\xd1:\xf5\x14\xee\x02\xc6\x0e\xee\x94J\xb84\x80\x8e\x94\x02zU\x0c\xed\x8d%$@R\xd00\x14\xb4\x00R\x8a@-)\x91W\xa9\xc5\x02!\x92\xf6(\xff\x00\x8a\xa9\xcd\xab\x81\x9d\xb4\x80\xa6\xfa\xb4\xb9\xe0\xd4gX\x98\x7f\x15\x03\x13\xfbn\x7fZ\x96\rnV\x90\x06\x14\x05\x8de\xb8\xdf\x18`*\'\x92c\xf7V\x90\x86\xa8\xba\'\xa0\xc5Y\x8d$#\xe6\x18\xa0c\xcc-Q\xb4/\xe9@\x10\xc9\x13c\xa1\xa8<\xb2O \xd2(r\xd9Fy5a`E\xe9\x8a\x02\xe3\xb6\x0e\xd4yt\x03#h\xf9\xa6\xf9t\xc4!\x8f\x9e\x94\xd3\x10\xf4\xa61\xa6\x15\xcfJi\x88zS\x01\xa6\x1ac\xdb\xabu\x14\x80\xae\xf6K\xd4U;\xa8\xb6\x8f\xa51\x10E\x07\x9b\xc084\xafi*v\xcf\xd2\x9d\xc6u\xe6\x8a\x08\nQ@\xc3\xebFh\x01\x19\xd5z\x9a\x82K\xd8\xd3\xbd )M\xaa\x7fv\xa9K\x7f#\xff\x00\x11\xa4\x05g\xb8\'\xa9\xa8ZRh\xb0\x0c.i\xb9\xaa\xb0\xec\x19\xa9\xady\x9dh`\xce\xc2\xc2%1\x0c\x8a\xb8"Q\xda\xb3\x10\xa1\x07\xa5;m\x00\x1bh\xdb@\r(=)\x86%\xfe\xe8\xa0c\x0c#\xd2\x9b\xe4\x8a\x06\x86\x98E4\xc6{\x1a\x02\xc3Lf\x82\xa6\x98\x11\xb1\xdb\xd6\x9b\xe6\'\xb50\x02P\x9aL)\xefL\x03`\xecE!\x8f\x8a\x064\xc7\xedY\xf7\xf0\x9d\xa4\xd0\x06u\xb1\xd9>\rm\xacj\xca\x08\xa0F\x85%Q\x01HX\x0e\xa6\x90\xc8\xa4\xba\x8d:\xb5R\x9bR\xc1\xc2\xd2\x02\x8c\xb7\xae\xfdZ\xab<\xc4\xf54\x01\x19\x93\xde\xa3,M\x03\x1b\x9aL\xd3\x18QL\x05\xa9m\xbf\xd7-&&vzo0\x8a\xbe\x05f!qF)\x8c\\Q\x8a\x04!\x14\xdd\xb4\x864\xad&\xda\x064\xad4\xad\x03cJ\xd3Yp)\x92R\xb88\x06\xa899\xa623#\x0e\xe6\x8f=\xc7z\x069n\x98\x1a\x99/Gz\x04N\xb7\x08\xc3\x86\xaa\x97\xf7!T\x8c\x03Lf+?\xcf\xb9kB\xd7T\xd8\x81]s@\x8d\xf2ED\xf3*u4\xc9*M\xa8*\xe7o5Bk\xf7~\x86\x80*\xbc\xcc\xc7\x93Q4\x94\x860\xb14\xdc\xd0\x02f\x92\x98\xc2\x8a`\x14\xb8\xa4\x03\x91rk[ [... cut ...] JJ\x00CI@\tE\x03\x10\x9aJ\x00JJ\x00CILBQHFX\x14\xe1Y\x10<R\x8ae\x0e\x14\xe1@\x0bK@\xc5\x14\xeac\x16\x81L\x05\x14\xb4\x0cZQ\xd6\x80%Z}P\x05-\x02\n3@\x06y\xa34\x00f\x8c\xd0\x01\x9ei(\x01(\xcd\x03\nC@\tHh\x10\x94\x1a\x06%!\xa6!);R$\xcd\xa5\x1dk"P\xb9\x14\xe0h\x18\xea\\\xd3\x18\xa0\xd2\xe6\x81\x8e\xcd.i\x80\xb9\xa5\xcd\x03\n\\\xd3\x01s\xc58\x1a\x10\xc9T\xd3\xb3V\x02\xe6\x8c\xd0\x01\x9a3@\x06h\xcd\x02\x0c\xd1@\xc34f\x80\x1b\x9a3H\x00\xb5&h\x013A4\xc0L\xd2f\x81\t\x9aBh\x10\x84\xd1\x9a\x04\x7f\xff\xd9'

3liasCo, 2025-05-30:
Coordenadas en centímetros de las esquinas de la hoja 
Esquina 
Inferior izquierda: (31 , 10.5) 
Superior izquierda: ( 50.8 , 10.5) 
Inferior derecha: ( 31 , -10.5) 
Superior derecha: (50.8 , -10.5) 

https://github.com/user-attachments/assets/528d8f51-11b6-45e2-8636-ace589ed7829

sebalander, 2025-05-30:
¡Buenísimo! Lo que te queda hacer es lo siguiente:

1. Abrí la imagen en algún programa que te permita identificar las coordenadas de las esquinas de la hoja (por ejemplo, el viejo Paint de Windows).
2. Anotá esas coordenadas directamente en este issue.
3. Luego, usá las mismas funciones que utilizaste antes para calcular la matriz de transformada.

Traé esto listo para el martes, así ese día podemos empezar directamente a usar esa matriz.

3liasCo, 2025-06-03:
MATRIZ DE TRANSFORMACION 

[[-2.18880937e-03 -2.63497709e-01 -2.44260579e+02]
 [ 1.02845724e+00  1.44625829e-03 -1.92956221e+02]
 [ 4.01073113e-04 -5.44887132e-02  1.00000000e+00]]

sebalander, 2025-06-04:
[draft] buenisimo.

sebalander, 2025-06-04:
@3liasCo necesitarias alguna ayuda o tutorial de python? te imaginas como habria que seguir? yo tengo algunas ideas. pero decime vos primero que te imaginas que habria que hacer y veo como guiarte.

3liasCo, 2025-06-06:
Me falta calibrar la matriz de transformación para el ángulo fijo en el que este la camara y fijarme si funciona y ¿necesitan las coordenadas reales de la pelota? O la distancia en si del robot a la pelota, no en forma de coordenadas, sino como el módulo del vector de esas coordenadas. 
Además de eso no sé que otra cosa más tendría que hacer con la open MV

3liasCo, 2025-06-06:
[image] 

Lo escribí medio mal pero espero que se entienda. 
También se puede calcular usando razones trigonométricas pero creo que medir el ángulo de la camara es menos preciso que medir su altura.

sebalander, 2025-06-09:
es una buena solucion. ta bueno. con las razones trigonometricas queda bien elegante. la solucion que vos ya mostras ahi:

d = D * (h - r) / h

permite deducir (te lo dejo como ejercicio jeje pero no es obvio) que las coordenadas x,y que queremos se calculan haciendo:

x = X * (h - r) / h
y = Y * (h - r) / h

donde X,Y son las coordenadas mal calculadas proproyectas hasta el plano del suelo, mientras que x,y son las coordenadas que queremos, proyectando hasta el plano del centro de la pelota

sebalander, 2025-06-11:
@3liasCo viendo entonces el comit que hiciste ayer: https://github.com/IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025/commit/c79d6f0afd905563eb1dea7996f27628bd089a5f
Esta bueno, es un paso importante. 

te dejo como tarea:
1. revisar que el calculo de la transformada de perspectiva es coherente con lo que hace OpenCV. es decir, lo que chatgpt nos dio al definir `aplicar_homografia(x, y, H)`  hay que ver que sea coherente con la definicion que tiene OpenCV de lo que es la transfromada de perspectiva proque es de OpenCV de donde sacamos nuestra matriz calibrada. Tenes para estudiar esta documentacion:
  - https://docs.opencv.org/4.11.0/d2/de8/group__core__array.html#gad327659ac03e5fd6894b90025e6900a7
  - https://docs.opencv.org/4.11.0/da/d54/group__imgproc__transform.html#ga20f62aa3235d869c9956436c870893ae
2. sacar del script cosas que ya no hagan falta, que hayan quedado de la forma anterior de calcular la distancia de la pelota, etc. 
3. > la cuenta de perspectiva tiene que ir despues de la homografia
   eso mismo, si

3liasCo, 2025-06-24:
https://colab.research.google.com/drive/1c7wPt0nK1N6JG70tcFwKNy3u0pTYOcno?usp=sharing

sebalander, 2025-06-25:
buenisimo el avance de hoy. que crees que te haria falta para ir haciendo para la semana que viene?

3liasCo, 2025-06-25:
> en el mensaje de error que copiaste no dice que linea del programa falla.

Me acuerdo que me marcaba en azul la línea 97 creo o alguna del final después de la línea 96

sebalander, 2025-06-25:
ok!! es una pista importante. o sea seguro que el problema estaen esta linea:

```python
    uart.write(bytearray(packet))            # Enviar por UART
```

proba de pasarle a alguna IA:
1. una frase diciendo que queres entender un error de codigo, bla bla, explicale que queres hacer
2. pegale el codigo entero para que lo vea completo
3. decile cual es el mensaje de error. pegale lo mas textual posible el mensaje de error y la linea del codigo conde creemos que esta el problema. 
4. pedile que te explique que es lo que causa el error

avisame si la explicacion que te de la IA se entiende y te ayuda a saber que hacer.

3liasCo, 2025-07-21:
Hola @sebalander quería saber si se podría hacer un espejo omnidireccional como en este video y saber (modificando el programa que ya está hecho) la distancia de la pelota, usando la transformación geométrica que ya estamos usando, y el ángulo de la pelota respecto del eje de la camara 

https://youtu.be/JBdiTaK9jbY?si=4rC_CfJzEvEQlEL2 

También encontré este video de unos rusos que utilizan un espejo omnidireccional para saber la posición de robot en la cancha aunque no dan muchos detalles 

https://youtu.be/om1k866kMfU?si=t2Oe7XyGBtXHBeGW

Fíjate en el minuto 4:22 tiene subtitulos

3liasCo, 2025-07-26:
<!DOCTYPE html>

<html>

<body>

distancia en pixeles al centro | distancia en centimetros al centro 
-- | --
24.0| 25
43.0| 50
66.0| 75
93.0| 100
118.0| 125
139.0| 150
156.0| 175
170.0| 200
182.0| 225
192.0| 250
201.0| 275
208.0| 300
31.11269837220809| 25√2
62.22539674441618| 50√2
98.99494936611666| 75√2
131.52186130069785| 100√2
156.97770542341354| 125√2
175.36248173426378| 150√2
190.91883092036784| 175√2
202.2325394193526| 200√2
212.13203435596427| 225√2 

</body>

</html>

sebalander, 2025-07-29:
[funcion de la camara omnidireccional.ggb.zip](https://github.com/user-attachments/files/21497597/funcion.de.la.camara.omnidireccional.ggb.zip)

descomprimilo y adentro esta el .ggb de geogebra tal como me lo pasate