IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #63: Comuniacion UART Arduino

Martii3141, 2025-05-17:
@Aczino593 te paso una documentacion que nos armo otro profe de como hicieron algunos chicos para poder tener una comunicacion entre la openMV y la Teensy. La propuesta es que vayas leyendo un poco, mas que nada en como recibe los datos la teensy, que es la placa que tiene la base movil que estaban soldando.
- Aca el link: https://github.com/IITA-Proyectos/WRO-SPORT-2025/issues/1#issuecomment-2887187106
Cualquier duda que tengas o si queres mas informacion, avisa asi te pasamos.

Aczino593, 2025-05-18:
Buenas profe pude hacer un código para que el maestro le envié un mensaje al esclavo. Pero como solo tengo una sola placa Arduino, la comunicación se realiza en una sola placa y no en 2.

[image]

sebalander, 2025-05-19:
@Aczino593 esta buenisimo si ya lo estas probando! esto es I2C? como hablas de maestro y esclavo pensamos que seria I2C. por lo que estuve leyendo UART tendria algunas ventajas respecto a I2C, por eso pregunto.

Aczino593, 2025-05-20:
@rafaelmoya2010 Fíjate la documentación que esta arriba sobre la comunicación OpenMV y teensy

sebalander, 2025-05-22:
# 📡 Protocolo de Comunicación UART: Envío de Pares `TAG–VALOR`

## 📜 Especificación Técnica del Protocolo

Se envían/reciben una secuencia ininterrumpida de bytes, uno atrás del otro sin parar.  
Este protocolo especifica cómo interpretarlos:

- El primer byte del par es un **TAG** (identifica qué dato se envía).  
- El segundo byte del par es un **VALOR** (codifica un número entre 0 y 200).  
- Siempre después de un byte de **TAG**, el que sigue debe ser sí o sí un byte de **VALOR**.

El mensaje consiste en una secuencia de pares `TAG–VALOR`.  
Puede tener cualquier cantidad de pares.

Ejemplo:
```
..., 201, 80, 202, 120, 203, 50, ...
```

---

### 🧾 Tabla de Rangos

| Byte  | Rango   | Significado                                  |
|-------|---------|----------------------------------------------|
| TAG   | 201–255 | Identifica el tipo de dato enviado           |
| VALOR | 0–200   | Información numérica asociada al tag previo |

---

### 🔖 Tabla de TAGs

| TAG | Nombre                        | Valor esperado | Conversión al valor real     | Descripción                                                  |
|-----|-------------------------------|----------------|-------------------------------|--------------------------------------------------------------|
| 201 | Pelota – coordenada X        | 0–200          | `x = v / 2`                   | Posición hacia el frente del robot (0–100 cm)               |
| 202 | Pelota – coordenada Y        | 0–200          | `y = v / 2 - 50`              | Posición izquierda/derecha (–50 a 50 cm)                    |
| 203 | Arco enemigo – coordenada X  | 0–200          | `x = v / 2`                   | Posición hacia el frente del robot (0–100 cm)               |
| 204 | Arco enemigo – coordenada Y  | 0–200          | `y = v / 2 - 50`              | Posición izquierda/derecha (–50 a 50 cm)                    |
| 205 | Arco propio – coordenada X   | 0–200          | `x = v / 2`                   | Posición hacia el frente del robot (0–100 cm)               |
| 206 | Arco propio – coordenada Y   | 0–200          | `y = v / 2 - 50`              | Posición izquierda/derecha (–50 a 50 cm)                    |

---

## 🧠 Ejemplo Interpretado

Mensaje recibido:
```
..., 201, 80, 202, 120, 203, 50, ...
```

Se interpreta como:

- TAG 201 → valor 80 → pelota en eje X → `x = 80 / 2 = 40 cm`
- TAG 202 → valor 120 → pelota en eje Y → `y = 120 / 2 - 50 = 10 cm`
- TAG 203 → valor 50 → arco enemigo en eje X → `x = 50 / 2 = 25 cm`

sebalander, 2025-05-22:
@Aczino593 fijate la especificaicon que acabo de poner arriba, anda implementandola para la teensy.

rafaelmoya2010, 2025-05-28:
Hoy lei la comunicación uart y empecé a crear el código para trasladar de open mv a teensy

sebalander, 2025-05-28:
buenisimo, trata de armar un borrador de ese codigo para el martes y traerlo para testearlo directamente. 

armalo como tarea para la casa y *pegalo en este issue* asi lo podemos ver antes del martes

rafaelmoya2010, 2025-06-03:
hoy en clases Aczino593 y yo probamos el código que hizo Aczino593 de micropython y teensy, nos daba los valores, pero los valores no cambiaban. Ósea que la OpenMV no mandaba los datos para que la Teensy los reciba. Posiblemente se deba a la conexión que hicimos, porque nos faltaron 2 cables.

Aczino593, 2025-06-10:
Código OpenMV
from pyb import UART
import time

uart = UART(3, 115200) #baudios 115200

# Variables simuladas (fake data)
x_map = 0        # Centro X mapeado (0 a 100)
ang = 0          # Ángulo mapeado (0 a 180)
sentido = 0      # Sentido (0: SV, 1: VI, 2: VD, 3: DV)

while True:
    # Actualizar valores de prueba
    x_map = (x_map + 10) % 101
    ang = (ang + 30) % 181
    sentido = (sentido + 1) % 4

    packet = [
        255, x_map,       # Marca de x_map
        254, ang,         # Marca de ang
        253, sentido      # Marca de sentido
    ]

    print("Enviando paquete:", packet)       # Mostrar qué se envía
    uart.write(bytearray(packet))            # Enviar por UART

    time.sleep_ms(500)  # Esperar medio segundo entre envíos

rafaelmoya2010, 2025-06-10:
en la clase de hoy yo y @Aczino593 logramos hacer la transferencia de datos de micropyton a teency con valores falsos
la próxima clase intentaremos hacer la transferencia con datos verdaderos

sebalander, 2025-07-04:
buenisimo! gran avance. 

@Aczino593 @3liasCo @rafaelmoya2010 me imagino que la clase que viene vamos a poder empezar a ver funcionando completo el sistema de vision. o sea me imagino algo como:

1. tener bien instalada la camara en el robot (con legos seguramente)
2. poner el robot en el origen de coordenadas de un sistema de coordenadas x,y en la mesa
3. poner a mano la pelota en alguna posicion x,y que podamos medir en cm
4. la camara deberia detectar la pelota, calcular las coordenadas, codificarlas, y mandar los bytes a la teensy
5. la teensy recibe los bytes, los decodifica y muestra por monitor serie las coordenadas  x,y de la pelota

[image]

en resumen la idea es que el sistema empiece a funcionar cada vez mas completo. 💪