IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #71: Diseñar cilindro del dribbler

sebalander, 2025-05-22:
pasame por whatsap el diseño del cilindro del dribbler. cuanto antes, asi lo puedo imprimir en la impresora 3D

sebalander, 2025-05-28:
lo que podes ir haciendo en la casa son dos cosas.

la primera es ponerle al diseño que ya tenes el triangulito de guia por adentro del agujero, todo a lo largo del agujero, para que el eje de lego no pueda girar.

lo segundo es un modelo que se parezca a esto:
[image]
un cilindro con radio mas chico en el medio.

sebalander, 2025-06-06:
@lauri-an tendrias que avanzar modificando el agujerito para el eje de legos tal que el eje no gire. como habiamos charlado. si lo podes cargar al STL en la carpeta que se llama "Diseño 3D" https://github.com/IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025/tree/main/Dise%C3%B1o3D yo trato de imprimirlo para el martes

lauri-an, 2025-06-17:
No se que debo hacer aparte de modificar el diseño 3d