IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #73: Calibracion de tiempo para movimiento lineal

sebalander, 2025-05-30:
@Vale15-tb en este issue por favor subi los datos que hayas tomado de la calibracion que estas haciendo para el movimiento lineal. o por lo menos hace una descripcion de lo que estas haciendo (como para documentar). por ejemplo, podes subir a este issue alguna foto del experimento explicando como se tiene que mover el robot y explicando que es lo que vas a medir.  

en algun momento como resultado de las mediciones vas a obtener alguna tablita de datos tipo 

| delay (mili segundos) | distancia recorrida (cm) |
|-----------------------|--------------------------|
| 500                   | ...                      |
| 600                   | ...                      |
| ...                   | ...                      |

o algo asi.

Vale15-tb, 2025-06-17:
[image]

sebalander, 2025-06-23:
@Vale15-tb estos datos que tenes en papel: pasalos en digital a una planilla de calculo.

Vale15-tb, 2025-06-24:
Tiempo (s)	Distancia (cm)
125	            7.60 
200	            12
250	            15.5
500	            30.5
1000	    59.7

Vale15-tb, 2025-06-25:
https://amyd.quimica.unam.mx/pluginfile.php/6119/mod_resource/content/2/4_TutorialRegresinLineal.pdf

Vale15-tb, 2025-07-01:
[

[RegresionLineal...2.0.xlsx](https://github.com/user-attachments/files/21008335/RegresionLineal.2.0.xlsx)

](url)