IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #78: Soporte para Motor amarillo

Martii3141, 2025-06-05:
@xio02 te propongo que busques de la pagina que te voy a dejar mas abajo, algún modelo que nos sirva de momento, para poder instalar de forma mas efectiva el motor amarillo, en todo el sistema. La idea es que vos elijas el que mas te convenza, descargues el archivo del modelo y nos pases el que termina en .STL, asi lo mandamos a imprimir.
Link de la pagina --> https://m.yeggi.com/q/yellow+motor+mount/
Tambien si querés buscar modelos que no sean de la pagina, podes googlear "yellow gearbox motor bracket" y por lo menos a mi fue la busqueda que mejores resultados me dió. Avisa cualquier cosa.

lauri-an, 2025-06-25:
Como?

lauri-an, 2025-06-25:
Ya vi uno pero no se como descargarlo

sebalander, 2025-07-23:
@Martii3141 te pido si podes imprimir a modo de testeo (en el slicer filling al 10% y espesor de corteza de 0.4mm, o lo que sea para que se imprima rapido no mas) de esta pieza de testeo.

https://github.com/IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025/blob/main/Dise%C3%B1o3D/soporte%20TT%20motor_recortado%202025-07-23%20-%20seba.stl

es para verificar que los orificios estan bien posicionadios. si es asi despues se puede mandar la pieza completa