IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #90: Videos de las Clases de arduino @3liasCo

sebalander, 2025-06-25:
@3liasCo   te paso los links de las clases de arduino para que las veas

https://youtu.be/H0KjoxP4cok

https://youtu.be/YSnZdQeCgxE

ademas de verlas, es esperable que tengas dudas o consultas, asi que escribinos por aca y te vamos a responder.

sebalander, 2025-07-10:
@3liasCo  viste los videos? los pudiste ver completos?

3liasCo, 2025-07-10:
Hola ya ví los vídeos me faltan los de python. El resto de los videos están en el mismo canal de Youtube? Ah y en caso de que lo necesitará conoces alguna página para usar la placa de Arduino, como un simulador de la placa de Arduino.

Martii3141, 2025-07-31:
@3liasCo te dejo aca un cursito de arduino de otro canal. Para practicar podes usar tinkercad, si no sabes como usarlo avisame y te paso algun video o te explico como se usa:
Playlist -> https://www.youtube.com/watch?v=40-xpC56-zQ&list=PLaOwYHXnu18u0p8888L_DiQNCE5Mcpp92&ab_channel=Cefuve