IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #95: Especificación de la Función "MovimientoGiro"

sebalander, 2025-06-25:
@vickyT049 la idea es documentar y discutior sobre ocmo definir la funciond emovimiento de giro. es muy parecido al trabajo del movimiento lineal. pero no entiendo bien, recuerdo haber visto que ya tenian mas o menos resuelta la cuestion del giro. que hace falta hacer todavia?

vickyT049, 2025-06-25:
> to @vickyT049 <https://github.com/vickyT049>.
> <IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025/issue/95/issue_event/18305136629
> @github.com>

Sii ya tenemos eso , los univo que falta es unorlo nada mas

Martii3141, 2025-07-11:
@vickyT049 este va a seguir siendo tu issue para la siguiente clase. Me gustaria que dejes en un comentario si es que pudistye avanzar con algo sobre esta logica que se estuvo charlando la ultima clase y si es que te falto subir algo de la clase tambien hacelo por aca.