IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #107: Movimiento con Giroscopo

Martii3141, 2025-07-24:
@Vale15-tb te dejo aca un codigo que indica las cosas que si o si van a tener que estar en el codigo que va a manejar el giroscopo. Y tambien una forma sencilla para que termines de completar vos la logica del codigo ya midiendo el angulo de movimiento en el eje Z y de esa forma el robot se pueda mvoer un angulo dado.
```cpp // --- 1. LIBRERIAS (¡ESTO ES OBLIGATORIO!) ---
#include <Wire.h>          // Para que Arduino hable con el sensor (comunicacion I2C).
#include <Adafruit_Sensor.h> // Una libreria base para sensores de Adafruit.
#include <Adafruit_BNO055.h> // La libreria especifica para tu sensor BNO055.
#include <utility/imumaths.h> // Ayuda con las matematicas internas del sensor.

// --- 2. EL SENSOR EN TU CODIGO (¡ESTO ES OBLIGATORIO!) ---
// Creamos un "objeto" llamado 'bno' para poder darle ordenes al sensor.
// El numero '55' es solo un identificador.
Adafruit_BNO055 bno = Adafruit_BNO055(55);

void setup(void) {
  // --- 3. PREPARAR EL MONITOR SERIE (¡ESTO ES OBLIGATORIO para ver los datos!) ---
  Serial.begin(9600);
  Serial.println("--- Iniciando Sensor de Giro (BNO055) ---");
  Serial.println("Mueve el sensor para que se calibre y veras los angulos.");
  Serial.println("------------------------------------------");

  // --- 4. INICIAR EL SENSOR (¡ESTO ES OBLIGATORIO!) ---
  // Le decimos al sensor que "despierte" y empiece a funcionar.
  // Si algo sale mal (por ejemplo, los cables estan flojos), te avisara.
  if (!bno.begin()) {
    Serial.println("¡Ups! No encontre el sensor BNO055. Revisa tus cables.");
    while (1); // Si no lo encuentra, el programa se detiene aqui.
  }

  delay(1000); // Esperamos un segundo para que el sensor este listo.

  Serial.println("Sensor listo. ¡A girar!");
  Serial.println("------------------------------------------");
}

void loop(void) {
  // --- 5. LEER LOS DATOS DEL SENSOR (¡ESTO ES OBLIGATORIO en cada vuelta del loop!) ---
  // Creamos un "paquete de datos" (lo llamamos 'datosSensor')
  sensors_event_t datosSensor;
  // Y le pedimos al sensor que llene ese paquete con la informacion mas reciente.
  bno.getEvent(&datosSensor);

  // --- ¡AQUI ESTA EL ANGULO QUE NECESITAS! ---
  // El angulo de giro en el eje Z (Heading/Yaw) es el que te dice hacia donde mira tu robot.
  // Es como la lectura de una brujula.
  float anguloDeGiroEjeZ = datosSensor.orientation.x;

  // --- AQUI ES DONDE USARAS 'anguloDeGiroEjeZ' ---
  // Por ejemplo, para hacer que tu robot gire a un angulo especifico:
  // if (anguloDeGiroEjeZ > 90) {
  //   // Mueve el motor para girar a la izquierda
  // } else if (anguloDeGiroEjeZ < 90) {
  //   // Mueve el motor para girar a la derecha
  // } else {
  //   // Deten el motor, ¡llegaste al angulo!
  // }

  // --- MOSTRAR EL ANGULO EN LA PANTALLA (Monitor Serie) ---
  Serial.print("Angulo de Giro (Eje Z): ");
  Serial.print(anguloDeGiroEjeZ, 2); // Muestra el numero con 2 decimales
  Serial.println(" grados"); // Agrega " grados" y salta a la siguiente linea

  delay(100); // Una pequeña pausa para no llenar la pantalla de numeros muy rapido.
}```

La idea es que trates de completar la logica del movimiento de los motores que irian dentro de los if. Anda completando esto en tu casa, asi cuando llegues el martes que viene al IITA lo probas de una y vamos modificando sobre eso que ya tenes escrito y pensado. Si tenes alguna duda avisa y te damos una mano

Martii3141, 2025-07-31:
@Vale15-tb subi por favor el codigo que hiciste ya sea aca o como codigo en la carpeta de movilidad asi vemos que hiciste

Vale15-tb, 2025-08-19:
#include <Arduino.h>
#include <zirconLib.h>
#include <Wire.h>          // Para que Arduino hable con el sensor (comunicacion I2C).
#include <Adafruit_Sensor.h> // Una libreria base para sensores de Adafruit.
#include <Adafruit_BNO055.h> // La libreria especifica para tu sensor BNO055.
#include <utility/imumaths.h> // Ayuda con las matematicas internas del sensor.

// --- 2. EL SENSOR EN TU CODIGO (¡ESTO ES OBLIGATORIO!) ---
// Creamos un "objeto" llamado 'bno' para poder darle ordenes al sensor.
// El numero '55' es solo un identificador.
Adafruit_BNO055 bno = Adafruit_BNO055(55);

unsigned long tiempoAnterior = 0;
const long intervalo = 1000; // 1 segundo

void setup(void) {
  InitializeZircon();
  // --- 3. PREPARAR EL MONITOR SERIE (¡ESTO ES OBLIGATORIO para ver los datos!) ---
  Serial.begin(9600);
  Serial.println("--- Iniciando Sensor de Giro (BNO055) ---");
  Serial.println("Mueve el sensor para que se calibre y veras los angulos.");
  Serial.println("------------------------------------------");

  // --- 4. INICIAR EL SENSOR (¡ESTO ES OBLIGATORIO!) ---
  // Le decimos al sensor que "despierte" y empiece a funcionar.
  // Si algo sale mal (por ejemplo, los cables estan flojos), te avisara.
  if (!bno.begin()) {
    Serial.println("¡Ups! No encontre el sensor BNO055. Revisa tus cables.");
    while (1); // Si no lo encuentra, el programa se detiene aqui.
  }

  delay(1000); // Esperamos un segundo para que el sensor este listo.

  Serial.println("Sensor listo. ¡A girar!");
  Serial.println("------------------------------------------");
}

void loop(void) {
  // --- 5. LEER LOS DATOS DEL SENSOR (¡ESTO ES OBLIGATORIO en cada vuelta del loop!) ---
  // Creamos un "paquete de datos" (lo llamamos 'datosSensor')
  sensors_event_t datosSensor;
  // Y le pedimos al sensor que llene ese paquete con la informacion mas reciente.
  bno.getEvent(&datosSensor);

  // --- ¡AQUI ESTA EL ANGULO QUE NECESITAS! ---
  // El angulo de giro en el eje Z (Heading/Yaw) es el que te dice hacia donde mira tu robot.
  // Es como la lectura de una brujula.
  float anguloDeGiroEjeZ = datosSensor.orientation.x;

  // --- AQUI ES DONDE USARAS 'anguloDeGiroEjeZ' ---
  // Por ejemplo, para hacer que tu robot gire a un angulo especifico:

      unsigned long tiempoActual = millis();
if (tiempoActual - tiempoAnterior >= intervalo) {
  Serial.print("tiempoActual ");
        tiempoAnterior = tiempoActual;
  Serial.print("tiempoActual ");      
}
 
   if (anguloDeGiroEjeZ > 90) {
    Serial.print("a la derecha ");
    
    motor1(anguloDeGiroEjeZ,0);
    motor2(anguloDeGiroEjeZ,0);
    motor3(anguloDeGiroEjeZ,0);
    delay(1000);
   
 }else{ 
   Serial.print(" a la izquierda");
    motor1(anguloDeGiroEjeZ,1);
    motor2(anguloDeGiroEjeZ,1);
    motor3(anguloDeGiroEjeZ,1);
    delay(1000);
   
  
   }
    
  //   // Mueve el motor para girar a la derecha
  // } else {
      //   motor1(0,1)
      //   motor2(0,1)
      //   motor3(0,1)
  //   // Deten el motor, ¡llegaste al angulo!
  // }

  // --- MOSTRAR EL ANGULO EN LA PANTALLA (Monitor Serie) ---
  Serial.print("Angulo de Giro (Eje Z): ");
  delay(1000);
  Serial.print(anguloDeGiroEjeZ, 2); // Muestra el numero con 2 decimales
  delay(1000);
  Serial.println(" grados"); // Agrega " grados" y salta a la siguiente linea

  delay(100); // Una pequeña pausa para no llenar la pantalla de numeros muy rapido.
}

Vale15-tb, 2025-10-04:
angulo sería igual a 
flotat angulo= anguloDeGiroEjeZ
Puede ser un número con coma

Vale15-tb, 2025-10-05:
int minVal=265; // Valor mínimo esperado de la lectura analógica
int maxVal=402; // Valor máximo esperado de la lectura analógica
double x, y, z; // Variables para almacenar ángulos calculados

void setup()
{
  pinMode(2, OUTPUT);   // Pin 2 configurado como salida (LED)
  pinMode(3, OUTPUT);   // Pin 3 configurado como salida (LED)
  pinMode(4, OUTPUT);   // Pin 4 configurado como salida (LED)
  Serial.begin(9600);   // Inicia comunicación serial a 9600 baudios
}

void loop()
{
  int vx = analogRead(A0);  // Lee valor analógico del pin A0
  int vy = analogRead(A1);  // Lee valor analógico del pin A1
  int vz = analogRead(A2);  // Lee valor analógico del pin A2

  // Mapea valores de lectura a un rango de -90 a 90 grados
  int xAng = map(vx, minVal, maxVal, -90, 90);
  int yAng = map(vy, minVal, maxVal, -90, 90);
  int zAng = map(vz, minVal, maxVal, -90, 90);

  // Calcula ángulos en grados usando funciones trigonométricas
  x = RAD_TO_DEG * (atan2(-yAng, -zAng) + PI);
  y = RAD_TO_DEG * (atan2(-xAng, -zAng) + PI);
  z = RAD_TO_DEG * (atan2(-yAng, -xAng) + PI);

  // Imprime valores de los ángulos en el monitor serial
  Serial.print("X: ");
  Serial.println(x);
  Serial.print("Y: ");
  Serial.println(y);
  Serial.print("Z: ");
  Serial.println(z);
  Serial.println("--------");

  delay(3000);  // Espera 3 segundos para la próxima lectura

  // Condicionales para encender LEDs si la lectura supera 5
  if (vx > 5) {
    digitalWrite(2, HIGH);  // Enciende LED en pin 2
    delay(3500);            // Mantiene encendido por 3.5 segundos
    digitalWrite(2, LOW);   // Apaga LED en pin 2
  }
  if (vy > 5) {
    digitalWrite(3, HIGH);  // Enciende LED en pin 3
    delay(3500);
    digitalWrite(3, LOW);
  }
  if (vz > 5) {
    digitalWrite(4, HIGH);  // Enciende LED en pin 4
    delay(3500);
    digitalWrite(4, LOW);
  }
}

Vale15-tb, 2025-10-05:
//código para probar si la función millis funciona....probar primero 
# include <Arduino.h>
#include <zirconLib.h>
angulo=60
tienpoAnterior=angulo
void setup(void)
{

  InitializeZircon();

}

void loop(void)
{   

unsigned long tiempoActual = millis(); 
if (tiempoActual - tiempoAnterior >= intervalo) {
Serial.print("tiempoActual ");
tiempoAnterior = tiempoActual;
Serial.print("tiempoActual ");
}
if (anguloDeGiroEjeZ > 90) {
Serial.print("a la derecha ");
motor1(100,0); 
motor2(100,0);
 motor3(100,0);
 delay(1000); 
}else{
Serial.print("a la izquierda");
motor1(100,1);
motor2(100,1);
motor3(100,1);
retraso(1000);