IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #112: Charlas de Diseño 3D

Martii3141, 2025-07-31:
@mijaelb78 @vickyT049 como les dije la ultima clase, estaria bueno que por algun medio charlen entre ustedes de como van a hacer sus diseños, asi los diseños no se pisan entre si. Si quieren hacer una videollamada en algun momento entre ustedes para charlar entre ustedes este tipo de cosas, tambien es opcion. Si no por chat o por los mismos comentarios de este issue pueden ir diciendo lo que estan pensando en hacer y llegar a un acuerdo entre algunas cosas que seguro que van a aparecer en el medio. 
Como recomendacion, si es que ya es algo mas que una idea y la estan ejecutando, tener las medidas a mano, alguna imagen, el mismo diseño en si, etc, para que de esa forma sea mucho mas visual para los dos.

mijaelb78, 2025-09-02:
@vickyT049 holaaa viki ya te subo la imagen  de la medida maxima de los pilares

mijaelb78, 2025-09-02:
[image]