IITA-Proyectos/RoboCupJunior-Soccer-Open-League-2025 #116: Investigación de los sensores infrarrojos Zircon

Aczino593, 2025-08-06:
Buenas profe. Necesito que me diga cual es el modelo de los 3 sensores infrarrojos que tiene instalado la Zircon para poder leer y hacer un código para estos sensores.
@sebalander

sebalander, 2025-08-07:
este esquematico de la Teensy/Zircon https://cdn.shopify.com/s/files/1/0892/9381/6107/files/Zircon.pdf

solo los llama "line sensors" y estoy buscando y no hay mas detalle sobre el modelo especifico de esos sensores.

pero lo importante es que hay disponible ejemplos https://robomov.net/blogs/zircon-tutorial/part-14 

si aun eso no te alcanza, avisame y te busco mas. o tambien proba con alguna IA que te pase un codigo de ejemplo y lo discutimos por aca

Aczino593, 2025-09-09:
Base para modificar

#include <Arduino.h>
#include <zirconLib.h>

const int UMBRAL_BLANCO = 750; // ajustar con la cancha real
const int VELOCIDAD = 100;     // velocidad normal de movimiento
const int VELOCIDAD_CORR = 100; // velocidad durante corrección
const int TIEMPO_CORR = 400;   // tiempo para alejarse de la línea (ms)

void setup(void) {
  Serial.begin(9600);
  InitializeZircon();
}

void loop(void) {
  int s1 = readLine(1); // sensor lateral izquierdo
  int s2 = readLine(2); // sensor trasero
  int s3 = readLine(3); // sensor lateral derecho

  // --- Muestra valores para calibrar ---
  Serial.print("S1: "); Serial.print(s1);
  Serial.print("\tS2: "); Serial.print(s2);
  Serial.print("\tS3: "); Serial.print(s3);

  // --- Lógica de escape ---
  if (s1 > UMBRAL_BLANCO) {
    Serial.print("\t--> Línea blanca IZQUIERDA, giro DERECHA");
    motor1(VELOCIDAD_CORR, 1); // rueda derecha atrás
    motor3(VELOCIDAD_CORR, 1); // rueda izquierda adelante
    delay(TIEMPO_CORR);
  }
  else if (s3 > UMBRAL_BLANCO) {
    Serial.print("\t--> Línea blanca DERECHA, giro IZQUIERDA");
    motor1(VELOCIDAD_CORR, 0); // rueda derecha adelante
    motor3(VELOCIDAD_CORR, 0); // rueda izquierda atrás
    delay(TIEMPO_CORR);
  }
  else if (s2 > UMBRAL_BLANCO) {
    Serial.print("\t--> Línea blanca ATRÁS, avanzo al frente");
    motor1(VELOCIDAD_CORR, 0);
    motor3(VELOCIDAD_CORR, 1);
    delay(TIEMPO_CORR);
  }

  // --- Movimiento normal (seguir jugando) ---
  motor1(VELOCIDAD, 0);
  motor3(VELOCIDAD, 1);

  Serial.println();
  delay(50); // pequeño delay para estabilidad
}

Aczino593, 2025-09-16:
Base a probar:

#include <Arduino.h>
#include <zirconLib.h>

const int UMBRAL_BLANCO = 750;   // Línea blanca
const int UMBRAL_VERDE_MIN = 300; //  verde (ajustar según valores reales)
const int UMBRAL_VERDE_MAX = 700;

int direccion = 1; // 1 = derecha, 0 = izquierda

void setup(void) {
  Serial.begin(9600);
  InitializeZircon();
}

void loop(void) {
  int s1 = readLine(1); // Izquierda
  int s2 = readLine(2); // Centro
  int s3 = readLine(3); // Derecha

  Serial.print("S1: "); Serial.print(s1);
  Serial.print("\tS2: "); Serial.print(s2);
  Serial.print("\tS3: "); Serial.print(s3);

  bool blanco_izq = s1 > UMBRAL_BLANCO;
  bool blanco_der = s3 > UMBRAL_BLANCO;

  bool verde_izq = s1 >= UMBRAL_VERDE_MIN && s1 <= UMBRAL_VERDE_MAX;
  bool verde_cen = s2 >= UMBRAL_VERDE_MIN && s2 <= UMBRAL_VERDE_MAX;
  bool verde_der = s3 >= UMBRAL_VERDE_MIN && s3 <= UMBRAL_VERDE_MAX;

  if (blanco_izq && !blanco_der) {
    // cerca del borde izquierdo, girar a la derecha
    direccion = 1;
    motor1(100, 1); // Motor izquierdo atrás
    motor3(100, 0); // Motor derecho adelante
    Serial.print("\tGirando a la derecha");
  } 
  else if (blanco_der && !blanco_izq) {
    // Cerca del borde derecho, girar a la izquierda
    direccion = 0;
    motor1(100, 0); // Motor izquierdo adelante
    motor3(100, 1); // Motor derecho atrás
    Serial.print("\tGirando a la izquierda");
  }
  else if (verde_izq && verde_cen && verde_der) {
    // Dentro del área, seguir patrullando en la dirección actual
    if (direccion == 1) {
      motor1(100, 1); // Ir a la derecha
      motor3(100, 1);
      Serial.print("\tPatrullando derecha");
    } else {
      motor1(100, 0); // Ir a la izquierda
      motor3(100, 0);
      Serial.print("\tPatrullando izquierda");
    }
  } 
  else {
    // Caso inesperado (fuera del área, negro, etc.)
    motor1(0, 0);
    motor3(0, 0);
    Serial.print("\t[ALERTA] Fuera de zona segura");
  }

  Serial.println();
  delay(100);
}

Aczino593, 2025-09-16:
Medicion umbrales:

#include <Arduino.h>
#include <zirconLib.h>

const int UMBRAL_BLANCO = 750; // Línea blanca
const int UMBRAL_VERDE_MIN = 300; // verde (ajustar según valores reales)
const int UMBRAL_VERDE_MAX = 700;

int direccion = 1; // 1 = derecha, 0 = izquierda

void setup(void) {
Serial.begin(9600);
InitializeZircon();
}

void loop(void) {
int s1 = readLine(1); // Izquierda
int s2 = readLine(2); // Centro
int s3 = readLine(3); // Derecha

Serial.print("S1: "); Serial.print(s1);
//Serial.print("\tS2: "); Serial.print(s2);
//Serial.print("\tS3: "); Serial.print(s3);

//bool blanco_izq = s1 > UMBRAL_BLANCO;
//bool blanco_der = s3 > UMBRAL_BLANCO;
//
//bool verde_izq = s1 >= UMBRAL_VERDE_MIN && s1 <= UMBRAL_VERDE_MAX;
//bool verde_cen = s2 >= UMBRAL_VERDE_MIN && s2 <= UMBRAL_VERDE_MAX;
//bool verde_der = s3 >= UMBRAL_VERDE_MIN && s3 <= UMBRAL_VERDE_MAX;
//
//if (blanco_izq && !blanco_der) {
//// cerca del borde izquierdo, girar a la derecha
//direccion = 1;
//motor1(100, 1); // Motor izquierdo atrás
//motor3(100, 0); // Motor derecho adelante
//Serial.print("\tGirando a la derecha");
//}
//else if (blanco_der && !blanco_izq) {
//// Cerca del borde derecho, girar a la izquierda
//direccion = 0;
//motor1(100, 0); // Motor izquierdo adelante
//motor3(100, 1); // Motor derecho atrás
//Serial.print("\tGirando a la izquierda");
//}
//else if (verde_izq && verde_cen && verde_der) {
//// Dentro del área, seguir patrullando en la dirección actual
//if (direccion == 1) {
//motor1(100, 1); // Ir a la derecha
//motor3(100, 1);
//Serial.print("\tPatrullando derecha");
//} else {
//motor1(100, 0); // Ir a la izquierda
//motor3(100, 0);
//Serial.print("\tPatrullando izquierda");
//}
//}
//else {
//// Caso inesperado (fuera del área, negro, etc.)
//motor1(0, 0);
//motor3(0, 0);
//Serial.print("\t[ALERTA] Fuera de zona segura");
//}
//
//Serial.println();
//delay(100);
}

Aczino593, 2025-10-05:
Detector de linea del modo delantero:

#include <Arduino.h>
#include <zirconLib.h>

float mover(float centimetros, int eje) {
  float tiempo = abs(((0.0654 * centimetros) - 0.1785) * 1000);
   if (centimetros>=0){
    if (eje == 3) {
      motor1(100, 0);
      motor2(100, 1);
      delay(tiempo);
      motor1(0, 0);
      motor2(0, 1);
    } 
    else if (eje == 1) {
      motor3(100, 0);
      motor2(100, 1);
      delay(tiempo);
      motor3(0, 0);
      motor2(0, 1);
    } 
    else if (eje == 2) {
      motor3(100, 0);
      motor1(100, 1);
      delay(tiempo);
      motor3(0, 0);
      motor1(0, 1);
    }
   }
   else {
    if (eje == 3) {
      motor1(100, 1);
      motor2(100, 0);
      delay(tiempo);
      motor1(0, 1);
      motor2(0, 0);
    } 
    else if (eje == 1) {
      motor3(100, 1);
      motor2(100, 0);
      delay(tiempo);
      motor3(0, 1);
      motor2(0, 0);
    } 
    else if (eje == 2) {
      motor3(100, 1);
      motor1(100, 0);
      delay(tiempo);
      motor3(0, 1);
      motor1(0, 0);
    }
   }
  return tiempo;
}

float girar(float angulo) {
  float tiempo = abs(angulo * 50 / 47.0);

   if (angulo>=0){
    motor1(100, 1);
    motor2(100, 1);
    motor3(100, 1);
    delay(tiempo);
    motor1(0, 1);
    motor2(0, 1);
    motor3(0, 1);
   }
   else {
    motor1(100, 0);
    motor2(100, 0);
    motor3(100, 0);
    delay(tiempo);
    motor1(0, 0);
    motor2(0, 0);
    motor3(0, 0);
   }
  return tiempo;
}
// -----------------------------------------------------------------------------------------

const int VELOCIDAD_CRUISE = 70;   // velocidad para avanzar normalmente
const int TIEMPO_POST_GIRO = 120;  // ms (pequeña pausa tras girar antes de avanzar)

// --- Rangos de color o umbral
bool esBlancoS1(int v){ return (v >= 575) && (v <= 753); }
bool esVerdeS1 (int v){ return (v >= 210) && (v <= 340); }

bool esBlancoS2(int v){ return (v >= 494) && (v <= 762); }
bool esVerdeS2 (int v){ return (v >= 370) && (v <= 467); }

bool esBlancoS3(int v){ return (v >= 766) && (v <= 780); }
bool esVerdeS3 (int v){ return (v >= 254) && (v <= 342); }

void setup() {
  InitializeZircon();
  Serial.begin(115200);
  delay(200);
}

void loop() {
  // leer sensores
  int s1 = readLine(1); // izquierda
  int s2 = readLine(2); // centro
  int s3 = readLine(3); // derecha

  // imprimir para calibración
  Serial.print("S1:"); Serial.print(s1);
  Serial.print(" S2:"); Serial.print(s2);
  Serial.print(" S3:"); Serial.print(s3);

  bool blanco_s1 = esBlancoS1(s1);
  bool verde_s1  = esVerdeS1(s1);
  bool blanco_s2 = esBlancoS2(s2);
  bool verde_s2  = esVerdeS2(s2);
  bool blanco_s3 = esBlancoS3(s3);
  bool verde_s3  = esVerdeS3(s3);

  // --- Si un sensor ve blanco y los otros 2 verde -> ejecutar giro ---
  if (blanco_s1 && verde_s2 && verde_s3) {
    Serial.println("\tDetectado: BLANCO IZQUIERDA -> giro (alejarse)");
    girar(-45.0);   
    delay(TIEMPO_POST_GIRO);
  }
  else if (blanco_s3 && verde_s2 && verde_s1) {
    Serial.println("\tDetectado: BLANCO DERECHA -> giro (alejarse)");
    girar(45.0);
    delay(TIEMPO_POST_GIRO);
  }
  else if (blanco_s2 && verde_s1 && verde_s3) {
    Serial.println("\tDetectado: BLANCO CENTRO -> giro 180 (o retroceder)");
    // opción 1: girar 180
    girar(180.0);
    delay(TIEMPO_POST_GIRO);
  }
  else {
    // --- Caso normal: todo verde -> avanzar continuamente ---
    Serial.println("\tTodo verde -> avanzando");
    // Para avanzar continuo usamos las mismas llamadas que usa mover() en adelante.
    // Ajusta direcciones si tu mapping es distinto.
    motor1(VELOCIDAD_CRUISE, 0); // ejemplo: motor1 adelante
    motor2(VELOCIDAD_CRUISE, 1); // ejemplo: motor2 adelante
    motor3(VELOCIDAD_CRUISE, 0); // si tu robot usa 3 motores para avanzar, mantener consistencia
  }

  delay(80); // pequeño retardo para estabilidad
}